zinkwazi/traffic-pcb-sb #3: I2C lines pulled to 5v instead of 3.3v

jdbaptista, 2024-08-29:
Currently, the I2C lines are pulled up to 5v, which I don't think the esp32 GPIO lines can handle. The matrix driver ICs can operate anywhere from 2.7-5.5v on data input lines when Vcc = 5v, so I think we can just change the 5v pullup to 3.3v.

Also, the resistors on the matrix driver data lines don't match the typical application circuit in the [datasheet](https://www.lumissil.com/assets/pdf/core/IS31FL3741A_DS.pdf) and are duplicated for the shared SDA and SCL lines.

I'm writing this as an issue instead of just making a commit because I don't want to break anything that was done intentionally. Any feedback before I make the change?

aidan-9880, 2024-08-29:
They should be pulled to 3.3v, that's my mistake. Used the wrong power symbol on accident. The resistor values and number of them is fine, 3 4.7k ohm in parallel = about 1600 ohms, around where it needs to be according to https://www.ti.com/lit/an/slva689/slva689.pdf (page 3). The i2c tracks are relatively long = high capacitance, so I didn't want to go too high. 

I have also made a few minor changes since I uploaded those files, will update the kicad files tomorrow with the final version when I'm in the office.

aidan-9880, 2024-08-30:
Switched to one 2k ohm resistor on each line, pulled to 3.3v. Also pulled the other LED driver resistors (on interrupt and shutdown pins) to 3.3v.

closed 2024-08-31 by jdbaptista