zinkwazi/traffic-pcb-sb #11: LED hardware numbers aren't logical with respect to the datasheet

jdbaptista, 2024-10-06:
ROWS are mapped backward on the CS matrix pins, causing the numbering system to be revered with respect to the datasheet. COLUMNS are split into two blocks, making things more confusing. These hardware numbers will be mapped properly in software, however the numbering differences makes the process error-prone. If there is another revision of the board, I think the LED numbers should be revised to make more sense with the datasheet.

jdbaptista, 2024-10-06:
Numbering should also be increasing by 1 on the ROW side, and spill over on the COLUMN side.

jdbaptista, 2024-10-06:
Pinout is different for each matrix, they should be standardized.

zinkwazi, 2024-12-07:
You only need to generate the LED matrix map once so, while it's not ideal, we cab live with this and fix it on the next projsect - Bay Area Freeways, anyone :)

aidan-9880, 2024-12-17:
Only reason for this was to make routing the matrix easier. Would've gotten pretty messy if stuff was crossing over everywhere. Definitely possible to change but not without a lot of rerouting.

jdbaptista, 2025-01-28:
Closing because nothing is logical anyway and there is no quick way to fix this without breaking mappings. Definitely something to keep in mind for future projects if we decide to make similar boards.

closed 2025-01-28 by jdbaptista