zinkwazi/traffic-pcb-sb #27: Screen print tweaks

zinkwazi, 2024-11-25:
- Remove the pointer lines from the current labels.
- Move Outside Open's label closer to the star and make the star bigger :) 
- Add labels for:
  - Long Beach
  - Glendale
  - Goleta :) 
  - Malibu
  - Santa Monica
  - Ventura
  - Anaheim

Don't be too concerned about being super accurate on the labels, this is a pretty broad strokes map.

jdbaptista, 2024-12-03:
I'm not sure how the silkscreen for the roads was created nor how I can modify it, so I'll ask Aidan about if removing the vertical western avenue is possible easily. I will make a PR for the other stuff shortly

aidan-9880, 2024-12-17:
I removed Western Ave in 30108bec5a281ee872067c4caef2b5d870d5c923. Should be good to close this.

zinkwazi, 2024-12-17:
Nice!

closed 2024-12-17 by zinkwazi