zinkwazi/traffic-pcb-sb #28: Architectural code changes

zinkwazi, 2024-11-25:
As we plan for version 2, I want to suggest a new approach and features that will improve scalability, efficiency, and functionality. 

    **Server-Side Data Processing:**
        Replace the current board-side TomTom querying system with a server-side Python application.
        The server will handle API calls to TomTom, aggregate traffic data, and format it into a concise payload for each board.
        This reduces API call costs and eliminates the need for each board to have a unique TomTom API key.

    **Unique Board Identification:**
        Boards will present a unique ID during each data request.
        Use the ESP32’s MAC address or whatever is available

    **Optimized Data Transmission:**
        Instead of sending data for every LED, the server will provide a compact list of LED states (e.g., R, B, G) in a serialized format.
        Use efficient protocols such as MQTT or compressed JSON to reduce overhead.

    **Memory Optimization on ESP32:**
        Reduce storage needs on the ESP32 by:
            Offloading data logic to the server (e.g., traffic segment mapping).
            Storing only LED states and logic for rendering them.
        This frees up space to support additional LEDs. 

    **Boot Behavior and Refresh Cycle:**
        On boot, the board will display north-to-south traffic conditions automatically.
        Implement a 5-minute interval to refresh and switch to displaying south-to-north traffic conditions.

jdbaptista, 2024-12-06:
So I've thought of a few options for supporting LED animations. The simplest and probably quickest is to store the PCB coordinates of the LEDs and calculate the order in which the LEDs should be turned on based on distance from some animation curve. For example, for a flow from top left to bottom right a straight line can be defined by the equation $y = tan(\theta)x$, with the dot product a point to a vector defining the line giving a signed distance $D=-sin(\theta)x+cos(\theta)y$. The D value of all LEDs can then be sorted and given to the I2C gatekeeper in that order. Calculations can be done to find the D value of many other types of lines and curves using linear algebra, however that may be difficult.

An alternative that is more in line with an adjacency matrix and will probably allow for a bit more flexibility is to use a variable precision tree to store points. This kind of tree works by splitting the coordinate space into a few partitions (think 4) and splitting those partitions further until only one LED is contained in any node of the tree. This is efficient when there is a lot of sparsity on the grid. This method is much more difficult for arbitrary angle sweeping animations, but should be quite fast and easy for animations in line with the grid used to define the LED coordinates. Even then, there will be more work to implement this method than the one discussed above, which is just more math heavy for adding new animation curves.

One other option is to use linked lists for each row and column in the adjacency matrix, with each list containing which LEDs are in that row or column. This should be easier to implement than the variable precision tree, however takes up double the amount of space necessary to store the data. With the amount of space used to store this, animations can be precomputed using the first method and stored directly in the program.

zinkwazi, 2024-12-07:
I like the first option - distance based calculations on an animation curve, but there may be a computational limit on the ESP32 that we'd have to experiment with.

-     Flexibility: This method would support a wide range of animations, including waves, flows, and sweeping effects. It's adaptable to straight l;ines, diagonal waves, and even simple curves.

-     Simplicity: It's the easiest to implement compared to the quadtree or linked list methods.

-     Low Memory Footprint: Only requires storing the cooordinates of LEDs and performing calculations dynamically.

-     Pre computation: Sorting dynamically can be computationally expensive, precomputing the sorted order for common curves can help mitigate this.

zinkwazi, 2024-12-07:
Use a LUT to store each LED's segment association and coordinates.
This way you'll be able to do the animations because the matrix will have the adjacency.

leds = [
    {"id": 0, "segment": "I-5_North", "x": 10, "y": 20},
    {"id": 1, "segment": "I-5_North", "x": 12, "y": 22},
    {"id": 2, "segment": "I-405_South", "x": 50, "y": 60},
    {"id": 3, "segment": "NULL", "x": 52, "y": 60}, //no LED at this location
    # ... continue for all LEDs
]
The array may contain additional details but you get the idea.

jdbaptista, 2025-04-15:
Aside from unique board identification, this is complete! V2_0 is a very nice board so far :)

closed 2025-04-15 by jdbaptista