zinkwazi/traffic-pcb-sb #29: v2 Stretch goals

zinkwazi, 2024-11-25:
**Stretch Goal for V2**

    **Ambient Brightness Adjustment:**
        Integrate a photoresistor to measure ambient light.
        Adjust LED brightness dynamically based on the surrounding light to improve comfort in low-light environments.

    **Additional LEDs:**
        Add LEDs for the 10 freeway.
        Add LEDs for the 110 freeway.

jdbaptista, 2024-12-20:
Unfortunately, I think the additional LEDs will have to wait for V3 simply due to time constraints

jdbaptista, 2025-01-02:
I believe that adding a new matrix driver is possible and will not draw too much power. The USB-C specification provides a minimum of 1.5A if both the source and drain receptacles are USB-C receptacles. I'll assume this to be the maximum we can draw without any modification to the circuit (source: https://hackaday.com/2022/12/06/usb-c-introduction-for-hackers/). The article actually states that the circuit we have now is configured to draw 3A, but its better to be safe.

The current of each CS line on the matrix drivers outputs a peak current of `Iout(peak) = 18.93mA * (GCC / 256) * (SL / 256)` where GCC is the global current control register and SL is the scaling register for the LED. There are 39 CS lines on each driver and the maximum safe GCC that the LEDs can handle at SL = 256 is 0x80 (ie. 128). Thus the maximum draw of an individual driver should be a little over `I = 39 * 18.93mA * (128 / 256) = 369.135mA. I'm not sure how much current the IC itself pulls, but we can round up to 400mA to be safe. With 4 drivers, this is a total of 1.2A@5V = 6 watts, leaving 1.5 watts for the rest of the circuitry. The biggest draw from the rest of the board is the ESP32, which specifies a recommended power supply output current of 500mA at 3.3V (1.65 watts).

The total here is 7.65 watts, which is slightly over the maximum. However, most of the time only one of the three LEDs for a particular location is on at a time, meaning the 4 drivers should realistically be pulling slightly over 2 watts at any time. If we ensure in the code that the sum of SL of the three registers does not exceed (2/3) of the maximum, the power draw from the drivers should only be 4 watts, bringing the total draw of the ESP32 and LEDs to 5.65 watts. There should be enough power for the rest of the circuitry, especially if the LEDs are not blasting at the maximum current they can handle.

jdbaptista, 2025-01-03:
[image]
This is what I will be implementing for new matrix boundaries. I will try to load balance the matrices a little bit so that future additions can be routed from the closest matrix driver instead of only from the new matrix (which would require cross-boundary lines).

zinkwazi, 2025-01-04:
Nice, great idea!

zinkwazi, 2025-01-04:
If we get really ambitious in the future, we can switch to a 5W power supply like Adafruit does for driving up  to 25K LEDs:
https://www.adafruit.com/product/1466

jdbaptista, 2025-04-24:
Added dynamic brightness adjustment in v0.8.0 1fdbe1c25b60d4e45bcf0b8c9c31f349a9717b69. This is now complete!

closed 2025-04-24 by jdbaptista